OpenEMS/openems #2557: Missing reactive power measurements for goodwe gridmeter

janwach, 2024-02-29:
### Bug Report or Feature Request (mark with an `x`)
```
- [ x ] bug report -> please search issues before submitting
- [ x ] feature request
```

### Bug description or desired functionality.

** Missing reactive power measurements**

Hello everyone,

we are currently using a FENECON Home system with a goodwe inverter for a project. Trying to read the reative power exchanged with the grid we encounter the follow issue: the measurement is always 0.0. Using a dedicated measurement system we see that there is some reactive power. Looking at the code (io.openems.edge.goodwe/src/io/openems/edge/goodwe/gridmeter/GoodWeGridMeterImpl.java ) I noticed that it seems that the reactive power measurement only recieves dummy values?

Thank you in advance for any insights.

Best

Sn0w3y, 2024-02-29:
> FENECON Home system with a goodwe inverter for a project. this is not an openEMS error as it works @ fenecon and in openems you certainly have no Fenecon branded Inverter, which means, the Software is not the same on it.

sfeilmeier, 2024-03-03:
@janwach: If there is an issue with or you are missing a purchased feature with a FENECON system, please get in touch with FENECON support -> https://fenecon.de/kontakt/.

For OpenEMS discussions and feature requests, we recommend the OpenEMS Community Forum -> https://community.openems.io. In fact we plan to close the Github Issues feature, to have all OpenEMS discussions in one place.

I'll close the issue here. Thanks!

closed 2024-03-03 by sfeilmeier