OpenEMS/openems #2822: When using Odoo as metadata, after adding a new Edge, all edges stop displaying on the backend UI dashboard.

katsuya, 2024-09-30:
### Description

Whenever a new Edge is added, the bug mentioned in the title always occurs. At this time, checking http://localhost:4200/overview shows that the getEdges API outputs the error 'Unable to find edge with id [xxx]'. However, restarting the backend resolves the issue, and all edges, including the newly added one, are displayed on the dashboard.

Here is the bug details:

- When initializing the connection with PostgreSQL, all edges are loaded into the cache.  
  [InitializeEdgesWorker.java#L39-L52](https://github.com/OpenEMS/openems/blob/a479107eb12a93f7e2e047e9d6ee6aa31b0e4dfb/io.openems.backend.metadata.odoo/src/io/openems/backend/metadata/odoo/postgres/InitializeEdgesWorker.java#L39-L52)
- The `getEdges` API calls the `getPageDevice` method:  
  [MetadataOdoo.java#L578](https://github.com/OpenEMS/openems/blob/a479107eb12a93f7e2e047e9d6ee6aa31b0e4dfb/io.openems.backend/metadata/odoo/MetadataOdoo.java#L578)
- Next, `getPageDevice` calls `convertToEdgeMetadata` for each edge:  
  [MetadataOdoo.java#L596-L603](https://github.com/OpenEMS/openems/blob/a479107eb12a93f7e2e047e9d6ee6aa31b0e4dfb/io.openems.backend/metadata/odoo/MetadataOdoo.java#L596-L603)
- Since the newly added Edge is not in the cache, the error `throw new OpenemsException("Unable to find edge with id [" + edgeId + "]")` occurs.
- The cache is only updated during the initialization of the connection with PostgreSQL, so a restart is required after adding a new edge.

### Screenshots

[image]

### Operating System

macOS

### How to reproduce the Error?

Steps to Reproduce:

- Use Odoo as metadata.
- While the backend is running, add a new Edge.
- Launch the backend UI and navigate to `/overview`.
- None of the edges registered in Odoo are displayed.

Sn0w3y, 2024-09-30:
This is not an actual Issue but more of a "Feature" as it is not intended to add more edges in Production without Restarting. If you want you could see in my PR where i tried to fix this "Issue" !

https://github.com/OpenEMS/openems/pull/2505

Greetings

katsuya, 2024-10-01:
Thank you for your response. I understand that this should be addressed as a feature rather than a bug. I would like to implement this as a feature, so it would be helpful if you could let me know why https://github.com/OpenEMS/openems/pull/2505 was not merged and what the issues were.

Thank you!

Sn0w3y, 2024-10-01:
Hi, i re-opened the PR so we can work on it maybe:

https://github.com/OpenEMS/openems/pull/2827

Please close the Issue so we can see that is is already WIP

closed 2024-10-01 by katsuya

Sn0w3y, 2024-10-03:
@katsuya could you maybe check my Code in the PR and confirm it working ?