OpenEMS/openems #2906: Ocpp server fail to send commands to the charging stations when two connections are established

katsuya, 2024-12-03:
### Description

When two connections are established to the same `charge point A`, the following bug occurs, causing OpenEMS to fail in properly handling the previous connection:

[image]

- When `connection B` is established, and a `lostSession` is triggered in `activeEvcSessions` for `connection A`, it incorrectly affects `charge point A` as a whole. This results in `connection B` malfunctioning as well.

The following issues should be addressed (esolving the first issue alone might be sufficient):

1. When a new connection is established, the old connection must be correctly removed from `activeEvcSessions`.
  - https://github.com/OpenEMS/openems/blob/263377b6282b9e812c218b045831c2f990ee0c1b/io.openems.edge.evcs.ocpp.server/src/io/openems/edge/evcs/ocpp/server/MyJsonServer.java#L105
3. When the old connection is disconnected, it should ensure that the `lostSession` for the `charge point` referenced by the new connection is not invoked.
  - https://github.com/OpenEMS/openems/blob/263377b6282b9e812c218b045831c2f990ee0c1b/io.openems.edge.evcs.ocpp.server/src/io/openems/edge/evcs/ocpp/server/MyJsonServer.java#L117
5. Disconnection of the old connection should not remove the `charge point` from `ocppSessions` associated with the new connection.
  - https://github.com/OpenEMS/openems/blob/263377b6282b9e812c218b045831c2f990ee0c1b/io.openems.edge.evcs.ocpp.server/src/io/openems/edge/evcs/ocpp/server/MyJsonServer.java#L128

### Operating System

Ubuntu

### How to reproduce the Error?

1. Connect `charge point A` to the OCPP server.  
2. Establish another connection between `charge point A` and the OCPP server without disconnecting the first connection.  
3. Disconnect the first connection.

katsuya, 2024-12-03:
> @katsuya: OCPP is not actually used right now on our side, so this has no high priority ATM.

@sfeilmeier That's really unfortunate. We've started using it, so if we send a patch from our side, would you consider reviewing it?

sfeilmeier, 2024-12-04:
> @sfeilmeier That's really unfortunate. We've started using it, so if we send a patch from our side, would you consider reviewing it?

Sure. I just wanted to point out that we are currently not working on this issue. If there is a working and reviewed patch, I'll of course merge it.

katsuya, 2024-12-10:
I previously reported that there might be some odd behavior related to multithreading (https://github.com/OpenEMS/openems/pull/2908#issuecomment-2515850362). However, after testing over the weekend in an environment where the Wi-Fi connection was intermittently dropping, we confirmed that there are no other issues aside from the two we have already reported (this issue #2906 and #2909). All other problems were related to the OCPP client side.

After testing the patch this weekend, I have a few concerns and will comment on them individually. @Sn0w3y

katsuya, 2024-12-16:
@Sn0w3y I left a comment on the PR (https://github.com/OpenEMS/openems/pull/2908#pullrequestreview-2490459735). Could you take a look when you have time?