OpenEMS/openems #2920: Incorrect historic average values

DennisLevl, 2024-12-11:
### Description

We have identified a bug when querying the historic `_sum/EssActivePower` for one of our simulated edges.
The query incorrectly returns the average of all existing entries for the requested resolution (e.g., 15 minutes) without considering that OpenEMS persists a value only when it changes (or at least every five minutes).

**Sample**
09:30:00: 5 kW
09:30:01: 10 kW
09:30:05: 5 kW
Querying the average between 09:30:00 and 09:30:05 yields:
(5+10+5)/3 = 6.67 kW.

However, the correct average should account for the persistence of values over time: (5+10+10+10+10+5)/6 = 8.33 kW.

---

**Setup**
Branch: Current dev-branch
Environment: Backend with simulated edge
Database: InfluxDB v2.7.10

---

**Explaination**
The field queried is `_sum/EssActivePower`.

The issue is illustrated in the influx graph below:
- EssActivePower remains constant near 0 kW until 09:30 am (point 1).
- At 09:30 am, the power increases due to the PID filter, reaching 800,000 kW within 35 seconds (point 2).
- After stabilizing at 800,000 kW, it gradually decreases back to near 0 kW by ~09:33 am (point 3), also using the PID filter.

[image]

During this time range, 74 entries exist in InfluxDB:

2 entries before the power increase, with a 5-minute gap between them.
35 entries while the power increases to 800,000 kW.
35 entries while the power decreases back to near 0 kW.
2 entries after the decrease, again with a 5-minute gap.

Requesting historic data via Postman for this range returns a mean value of 377,539 kW. This matches the result obtained using the `mean ` function in InfluxDB.

[image]

Influx Query: 
```
from(bucket: "openems")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "data")
  |> filter(fn: (r) => r["_field"] == "_sum/EssActivePower")
  |> filter(fn: (r) => r["edge"] == "1")
```
  
  Postman Query:
  ```
{
    "method": "edgeRpc",
    "params": {
        "edgeId": "edge1",
        "payload": {
            "method": "queryHistoricTimeseriesData",
            "params": {
                "timezone": 0,
                "fromDate": "2024-11-15",
                "toDate": "2024-11-15",
                "channels": [
                    "_sum/EssActivePower"
                ],
                "resolution": {"value": 15,"unit": "Minutes"}
            }
        }
    }
}
```

---

**Expected behaviour**

When accounting for gaps using the last existing value to fill missing timestamps, the correct average would be 121,724 kW.

[image]

Adjusted Influx Query: 
```
from(bucket: "openems")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "data")
  |> filter(fn: (r) => r["_field"] == "_sum/EssActivePower")
  |> filter(fn: (r) => r["edge"] == "1")
    |> aggregateWindow(every: 1s, fn: last, createEmpty: true)
  |> fill(usePrevious: true)
  |> aggregateWindow(every: 15m, fn: mean)
```
  
This approach correctly considers the persistence of values over time, resulting in the expected average.

### How to reproduce the Error?

1. Setup an OpenEMS instance with Influx as time series database
2. Produce sample data for the field `_sum/EssActivePower` (e.g. due to a simulated production/consumption datasource with entries like 5, 10, 10, 10, 10, 5)
3. Query the configured time range with the history method `queryHistoricTimeseriesData` via postman. You could also check the amount of entries in Influx for this time range.

DennisLevl, 2024-12-12:
I see, thank you for your explanation. I labelled it as a bug because in multiple cases we need time-weighted averages in OpenEMS to check the history (e.g. the energy monitor in the UI). But I also understand your arguments.
Your solution to add a property as part of the params sounds great. From my point of view this would be a great feature to increase the accuracy of the data.

DennisLevl, 2024-12-13:
You're right. In this case, the gaps would still be filled with the previous value, which is also not a perfect solution.

Let me give you some background information on why we encountered this issue. In early November, we transitioned from release 2024.9.0 to the current dev version (as of November 6, 2024). Prior to this update, values like _sum/EssActivePower, _sum/GridActivePower, and _sum/EssSoc were stored in Influx every second, regardless of changes.

[image]

With the update, however, this behavior changed, and data is now only stored when there are changes, or at least every 5 minutes.

[image]

As we need to analyze the power usage of our batteries at various resolutions for accurate forecasting, it is essential to have technically precise data.

Ultimately, the solution with fill() was just a workaround for the underlying issue of missing data points.

Unfortunately, I haven't yet determined what caused this change. Since time-series databases like Influx are optimized for storing large amounts of data and use various compression algorithms, the advantage of consistent data for _sum-channels should outweigh the additional data storage.

From my perspective, at least for the past month, the data should be available in the corresponding resolution. There are a lot of use cases where high-resolution data plays an important role.

DennisLevl, 2024-12-19:
Thank you for your message. 
As can be seen in my first message, we are using the ‘edgeRpc’ method to query the historicTimeseriesData. As I understand it, this should forward the request directly to the edge: [Internal Component Communication :: Open Energy Management System](https://openems.github.io/openems.io/openems/latest/component-communication/index.html#_edge_rpc)

I have set up an Influx-DB on the edges, which shows that the data is saved with a resolution of every second. 
Information is only lost during synchronisation with the backend.

However, I still get the data from the backend. If you look at https://github.com/OpenEMS/openems/blob/develop/io.openems.backend.core/src/io/openems/backend/core/jsonrpcrequesthandler/EdgeRpcRequestHandler.java  lines 77 and 126, you can see that an edgeRpc call first attempts to load the data from the backend and only loads it from the edge if this is not possible. 
My expectation would have been the other way round? 

So I have the following questions and maybe someone can help me answer one of them: 
- Is this prioritisation (backend, then edge) intentional? If so, why? It seems unintuitive to me because of the naming. Also, the data in the edge is available in a higher resolution. It also places an unnecessary burden on the backend if it has to deal with the request itself instead of passing it on. Especially with a high number of devices.
- Let's assume I get the data from the Edge. Is there a regulation on how long the data is retained by the edge?
- In which commit was the resolution adjusted? I used to be able to retrieve the data from the backend in a higher resolution (we didn't even have a DB defined for the edges before)

Ultimately, I need some solution to retrieve the data in the correct resolution. Otherwise we can't calculate our predictions reliably.

sfeilmeier, 2025-03-20:
@DennisLevl: I briefly discussed this with @michaelgrill. The desired setup of OpenEMS Backend uses "Timedata.AggregatedInfluxDB", i.e. the data you receive is pre-aggregated at the Edge.

The actual code is here, and its true, that there is no weighted average. But at the same time locally there is usually one value per second available, so the problem should not exist.
https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.controller.api.backend/src/io/openems/edge/controller/api/backend/SendChannelValuesWorker.java#L206-L213

So if you query the data from a Backend that uses "Timedata.AggregatedInfluxDB", the problem should not exist. Are you sure it really does?

DennisLevl, 2025-03-25:
@sfeilmeier: Thank you very much for your response! Actually we only used the regular Timedata.InfluxDB bundle in our Backend. 
Like described in https://openems.github.io/openems.io/openems/latest/backend/service.adoc.d/io.openems.backend.timedata.aggregatedinflux.html I thought the AggregatedInfluxDB bundle is only for backends who managed a lot of edges.

Now, with the AggregatedInfluxDB bundle, the averages and the edge values actually match!

Two thoughts on this: 
1. We are already in exchange with several companies using an OpenEMS based EMS. We have observed this false averaging behaviour on some of their systems as well. It might be a good idea to adjust the documentation of the backend so that using the Timedata.InfluxDB only leads to wrong averages. 
2. Setting up the AggregatedInfluxDB was a bit tricky. You have to define a bucket (e.g. "aggregated0"), but actually the bucket name consists of the bucket name and the retention policy name for averages (e.g. "rp_avg"). This means that you have to actively create a bucket called "aggregated0/rp_avg", otherwise no data will be written. However, the configuration made me think that the bucket name should be "aggregated0" and the retention policy name is only used for InfluxDB v1 (we used InfluxDB v2).