OpenEMS/openems #3047: Zero feed-in activation

cf99, 2025-03-13:
### Component

Edge

### Description

In Austria the government is most probably introducing a new law. Currently everybody has to pay for the grid while obtaining energy. In future this might be also valid for providing surplus energy  to the grid! 
In a case where the grid-costs are higher than the price of the sold energy (to the grid) i want to switch to a zero feed-in mode. (grid costs: ~0,1€/kWh > sold energy: 0,05€/kWh)
In my case the price change every month (awattar)
During the installation the limit is set to a certain value that cant be changed by the end user. This restriction must be valid further.

cf99, 2025-03-13:
yes.... 

did i posted it wrong? (if so i am sorry, just wanted to raise a feature somehow)

sfeilmeier, 2025-03-13:
Hi @cf99,

yes, I think a Github Issue is the wrong place to discuss this. Lets move the discussion to OpenEMS Community and keep it in German language: https://community.openems.io/

Also this webinar might be interesting for you, regarding latest developments in FEMS (and OpenEMS) for new German legislation, which has some things in common with the law in Austria: https://www.youtube.com/watch?v=6JBXpeb1s5I

I am closing here. Thanks!

closed 2025-03-13 by sfeilmeier