OpenEMS/openems #3081: Connection error with modbus after upgrade

chakalaka91, 2025-04-01:
### Description

### The problem

Hey all,

I'm using the modbus integration to pull data from my Fenecon inverter, which has a modbus/TCP API.

Since setting up some years ago this worked just fine, however since yesterday it doesn't work anymore. 

The entities are not available anymore. Interestingly, the battery sensor "believes" it's working, however the provided charge state is not correct.

[image]

I've integrated the inverter via a separate yaml, which is called modbus.yaml and I reference to it in the configuration.yaml with 

```
# FEMS Modbus Configuration
modbus: !include modbus.yaml
```

I'd be glad if anyone could help me this, since I could'nt get behind, why this doesn't work anymore. To me the configuration seems correct.

Kind regards

### Example YAML snippet

```yaml
- name: "fems"
  delay: 0
  timeout: 5
  type: tcp
  host: 192.168.178.199
  port: 502
  
  sensors:
    #INT
    - name: "FEMS_EssSoc" # Battery SoC
      scan_interval: 60 
      data_type: uint16
      input_type: input
      device_class: battery
      state_class: measurement
      unit_of_measurement: "%"
      address: 302
      slave: 1
      unique_id: fems_modbus_302

    #FLOAT32      
    - name: "FEMS_EssActivePower" # Combined Power of PV + Battery
      unit_of_measurement: W
      scan_interval: 10
      data_type: float32
      input_type: holding
      device_class: power
      state_class: measurement
      address: 303
      slave: 1
      unique_id: fems_modbus_303

    - name: "FEMS_GridActivePower" # Grid Power
      unit_of_measurement: W
      scan_interval: 10 
      data_type: float32
      input_type: holding
      device_class: power
      state_class: measurement
      address: 315
      slave: 1
      unique_id: fems_modbus_315

    - name: "FEMS_ProductionActualPower" # PV Power
      unit_of_measurement: W
      scan_interval: 10
      data_type: float32
      input_type: holding
      device_class: power
      state_class: measurement
      address: 339
      slave: 1
      unique_id: fems_modbus_339
      
    - name: "FEMS_ConsumptionActivePower" #House Power
      unit_of_measurement: W
      scan_interval: 10
      data_type: float32
      input_type: holding
      device_class: power
      state_class: measurement
      address: 343
      slave: 1
      unique_id: fems_modbus_343
      
            
    - name: "FEMS_EssDischargePower"  #Battery Discharge Power
      unit_of_measurement: W
      scan_interval: 5 
      data_type: float32
      input_type: holding
      device_class: power
      state_class: measurement
      address: 415
      slave: 1
      unique_id: fems_modbus_415
      
    #Energy  (Float64)
      
    - name: "FEMS_GridBuyActiveEnergy" #Total Consumption from Grid
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 359
      slave: 1
      unique_id: fems_modbus_359 
      
    - name: "FEMS_GridSellActiveEnergy" #Total Energy Sold to Grid
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 363
      slave: 1
      unique_id: fems_modbus_363

    - name: "FEMS_ProductionActiveEnergy" # Total PV Production Energy
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 367
      slave: 1
      unique_id: fems_modbus_367
      
    - name: "FEMS_ConsumptionActiveEnergy" # Total Energy Usage
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 379
      slave: 1
      unique_id: fems_modbus_379
      
    - name: "FEMS_EssDcChargeEnergy" # Total Battery Charge Energy
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 383
      slave: 1
      unique_id: fems_modbus_383
      
    - name: "FEMS_EssDcDischargeEnergy" # Total Battery Discharge Energy
      unit_of_measurement: Wh
      scan_interval: 60 
      data_type: float64
      input_type: holding
      device_class: energy
      state_class: total_increasing
      address: 387
      slave: 1
      unique_id: fems_modbus_387
```

### Anything in the logs that might be useful for us?

```txt

```

### Operating System

Home Assistant Core 2025.3.4

### How to reproduce the Error?

An excerpt of the debug protocol gave me this: 
_2025-03-31 13:54:20.757 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.199', 502)
2025-03-31 13:54:22.050 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: fems: Error: device: 1 address: 415 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.199:502]
2025-03-31 13:54:25.961 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-03-31 13:54:31.365 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-03-31 13:54:37.167 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-03-31 13:54:38.776 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.199', 502)
2025-03-31 13:54:41.982 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.199', 502)
2025-03-31 13:54:48.389 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.199', 502)
2025-03-31 13:55:01.195 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.199', 502)
2025-03-31 13:58:42.095 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x89 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0x9f 0x0 0x2
2025-03-31 13:58:42.101 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x89 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xff 0xff 0xff 0xff old_data:  addr=None
2025-03-31 13:58:42.102 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x89 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xff 0xff 0xff 0xff
2025-03-31 13:58:42.102 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[65535, 65535], status=1) 
2025-03-31 13:58:42.102 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-03-31 13:58:43.127 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x8a 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0x2f 0x0 0x2
2025-03-31 13:58:43.130 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x8a 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xff 0xff 0xff 0xff old_data:  addr=None
2025-03-31 13:58:43.130 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x8a 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xff 0xff 0xff 0xff
2025-03-31 13:58:43.131 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[65535, 65535], status=1)_ 

It then tries to reset the header several more times.

chakalaka91, 2025-04-01:
> Hi, since others did not Report any errors - could you please report this to Home Assistant?
> 
> This is not an OpenEMS Issue i guess.
> 
> Greetings

That was the first place, where I reported. However I now know, that it immediatly stopped working, when I've upgraded to 2025.3.2. So something in the modbus configuration did change. I'm actually curious why not more people seem to have issues with that.

chakalaka91, 2025-04-01:
> > > Hi, since others did not Report any errors - could you please report this to Home Assistant?
> > > This is not an OpenEMS Issue i guess.
> > > Greetings
> > 
> > 
> > That was the first place, where I reported. However I now know, that it immediatly stopped working, when I've upgraded to 2025.3.2. So something in the modbus configuration did change. I'm actually curious why not more people seem to have issues with that.
> 
> Please upgrade to 2025.4.1 :)

Yes, just did. No changes unfortunately.

fabian94533, 2025-04-02:
Hi @chakalaka91I have also noticed this issue and am currently working on a solution. I found that the Modbus TCP API does not generate the Modbus table in certain cases, particularly after updates and sometimes even after regular restarts. I'll investigate this further.

As a temporary workaround, you need to modify the component. For example, you can change the port, immediately revert it, and then click "Update Component." The key is to trigger the component's modified method.

chakalaka91, 2025-04-02:
Hi @fabian94533 @Sn0w3y,

Since I was only grabbing the data provided by modbus, I didn't fumble around with the interface on FEMS side. I noticed some days ago, that the table was indeed empty. I thought that would be weird, maybe the data would be provided somewhere else?
**However** since the upgrade yesterday night I restarted the FEMS several times and that did actually help. The table has entries now and my Home Assistant receives data from FEMS again.

Thanks for your responses and willingness to help.

closed 2025-04-02 by chakalaka91

fabian94533, 2025-04-02:
@chakalaka91 I assume this error is due OSGi but i have not figured out why by now. There is no pattern behind the table not getting filled nor can i simulate this error. Anyways I am working on a fix for this. I am glad that restarting had helped you. I do my best to get this fixed until the next release.

benniju, 2025-04-02:
I also have the same issue. It also works for me after rebooting multiple times.
Would it be an option / temporary workarourd to uninstall the App in the App-Center? Can you simply reinstall the App afterwards?

merlinste, 2025-04-03:
De-installing the Modbus App did work for me!
At first, I was asked for a license key (which I dont have, as this app is free of charge), but after some minutes it seemed to re-install by itself, and now Modbus works again.

fabian94533, 2025-04-03:
@benniju @merlinste apparently, I wasn’t aware that normal users cannot enable or disable components. The main requirement is simply to trigger the component’s modified method to ensure it reloads correctly. Another way to achieve this would be to change the alias, for example, and then click "Update."

Anyway, I now have a better understanding of how this error occurs. It seems that some OSGi @Reference annotations are not populated in time after restarting the EMS. This issue occurs most frequently right after a firmware update but can also happen during normal restarts. I’m still working on resolving this problem.