OpenEMS/openems #3112: UI showing blank website

nanoBit93, 2025-05-01:
### Description

Hey, 
I'm new to openEMS and was trying to install it the first time (i'm using it in docker running on a Proxmox VM). I was using the docker-compose shown [here](https://github.com/OpenEMS/openems/tree/develop/tools/docker/edge) and it seems to work fine on the server side (`docker ps` showing edge and ui are up and running). 

However, when I try to access the UI, it only shows a blank page. There's an error message in the Chrome developer tools:
`NullInjectorError: No provider for NavigationService!` which seems to be the cause of having only a blank website. Complete Chrome Error:
`myerrorhandler.ts:15 NullInjectorError: R3InjectorError(AppModule)[ApplicationInitStatus -> InjectionToken Application Initializer -> [object Object] -> NavigationService -> NavigationService]: 
  NullInjectorError: No provider for NavigationService!
    at NullInjector.get (core.mjs:1621:27)
    at R3Injector.get (core.mjs:2148:33)
    at R3Injector.get (core.mjs:2148:33)
    at injectInjectorOnly (core.mjs:1116:32)
    at ɵɵinject (core.mjs:1122:60)
    at injectArgs (core.mjs:1253:23)
    at Object.factory (core.mjs:2374:52)
    at R3Injector.hydrate (core.mjs:2276:43)
    at R3Injector.get (core.mjs:2138:33)
    at injectInjectorOnly (core.mjs:1116:32)`

Does someone know how to fix it?

### Operating System

Ubuntu 24.04.2

### How to reproduce the Error?

1) install edge and ui via docker-compose from [this website](https://github.com/OpenEMS/openems/tree/develop/tools/docker/edge)
2) access ui through http:<ip of the VM>

sfeilmeier, 2025-05-01:
Hi,

It seems we currently have problems with our Docker build. I am not an expert, but as the bug you are describing is fixed in latest  develop-branch, maybe it should work to switch to `develop` tag https://hub.docker.com/r/openems/edge/tags.

You are welcome to join our community at https://community.openems.io/

Regards,
Stefan

closed 2025-05-01 by sfeilmeier

nanoBit93, 2025-05-01:
I tried using the develop tag now, but the same issue occurs.

nanoBit93, 2025-05-01:
Hey, 
thank you for the suggestion. I tried using the develop tag and I'm still facing the same issue.

Best regards,
Julian

sfeilmeier, 2025-05-02:
@da-Kai: Docker-Builds are currently broken. Could you maybe have a look?