OpenEMS/openems #3156: UI empty with multiple errors

lucabmn, 2025-05-30:
### Description

Hi, 

I tried to install openems via docker, here is my docker compose file:

```version: "3"

services:
  openems-edge:
    image: openems/edge:develop
    container_name: openems_edge
    hostname: openems_edge
    restart: unless-stopped
    volumes:
      - openems-edge-conf:/var/opt/openems/config:rw
      - openems-edge-data:/var/opt/openems/data:rw
    ports:
      - 8080:8080 # Apache-Felix
      - 8085:8085 # UI-Websocket

  openems-ui:
    image: openems/ui-edge:develop
    container_name: openems_ui
    hostname: openems_ui
    restart: unless-stopped
    volumes:
      - openems-ui-conf:/etc/nginx:rw
      - openems-ui-log:/var/log/nginx:rw
    environment:
      - UI_WEBSOCKET=ws://localhost:8085
    ports:
      - 880:80
      - 8443:443

volumes:
  openems-edge-conf:
  openems-edge-data:
  openems-ui-conf:
  openems-ui-log:```

but when I try to open the UI i recive multiple errors:

```NullInjectorError: R3InjectorError(AppModule)[ApplicationInitStatus -> InjectionToken Application Initializer -> [object Object] -> NavigationService -> NavigationService]: 
  NullInjectorError: No provider for NavigationService!```

```TypeError: this.service.websocket is null```

But there is no error in the UI console.

```Error: Promised response from onMessage listener went out of scope```

In the console of the edge container I recive this error:

```s6-rc-oneshot-run: fatal: unable to exec /etc/s6-overlay/s6-rc.d/init-adduser/run: Permission denied

s6-rc: warning: unable to start service init-adduser: command exited 126```

I already tried different versions, even the develop version.

### How to reproduce the Error?

Install OpenEMS with the provided docker compose.

lucabmn, 2025-05-30:
already tried the develop version

thatguy7, 2025-06-03:
I tried running recent versions of OpenEMS and encoutered the same problem as @lucabmn. When starting the edge container I get the error.

```
s6-rc-oneshot-run: fatal: unable to exec /etc/s6-overlay/s6-rc.d/init-adduser/run: Permission denied
s6-rc: warning: unable to start service init-adduser: command exited 126
```

The problem is present in most recent release(s). My workaround it to run an older release, specifically `openems/edge:2025.3.0` and `openems/ui-edge:2025.3.0`. Though I did not verify if it is the last version without the error.

fritzrichter, 2025-06-03:
Same here, latest does not work

PR #3166 (2025-06-04): Docker: set execute permissions to 'run'-files
fix #3156

> s6-rc-oneshot-run: fatal: unable to exec /etc/s6-overlay/s6-rc.d/init-adduser/run: Permission denied
> s6-rc: warning: unable to start service init-adduser: command exited 126