OpenEMS/openems #3169: Missing response to getSystemUpdateState requests on OpenEMS systems

Lamarqe, 2025-06-05:
### Description

I noticed that getSystemUpdateState requests remain unresponded in my local OpenEMS test system.

I analyzed and found that the following code
https://github.com/OpenEMS/openems/blob/ab5b5d2258aa009375187b905b35373fd4ebc66d/io.openems.edge.core/src/io/openems/edge/core/host/SystemUpdateHandler.java#L83
calls an exception handler without providing an exception (ex == null).

I added a proper IOException. This resolves the issue and the requests are properly responded (with an error response)

### Operating System

Ubuntu 24.04

### How to reproduce the Error?

Navigate in the WebUI to settings -> systems. 

Trace the websocket communication in the browser console.

Lamarqe, 2025-06-06:
dpkg is installed. But there is no configuration in openems which defines update parameters like download url or package name. All these parameters are null. So the dpkg command line contains a null package name,  which it apparently cannot find.

Where does the update information come from in your environment? And how does an openems debian package get built which can be updated via the debian package management?

sfeilmeier, 2025-08-03:
Fixed in #3170

closed 2025-08-03 by sfeilmeier