OpenEMS/openems #3202: UI: Add additional info to Meter Type in create component drop-down

sjjh, 2025-06-20:
### Component

UI

### Description

If I create a new "IO Shelly Pro 3EM" component in the UI, I get a drop-down dialog to choose the meter type (cf.screenshot below). Each type has a name, but not more information is shown to the user. But at the same time the javadoc of the meter type offers more information, as seen in https://github.com/OpenEMS/openems/blob/develop/io.openems.common/src/io/openems/common/types/MeterType.java IMHO it would be helpful to the user to include this additional information to the dialog. For example, in this case the context information about the differences between *consumption metered* and *consumption non metered* and the consequences for *calculating consumption* initially wasn't  clear to me.
I can imagine different ways in doing so (all having some downsides):
- add description text to existing name label in drop-down dialog (could clutter the drop-down)
- add description text to existing name label in drop-down dialog, but hide behind a "more" label/accordion (adds click/touch event to an item already having an click/touch event)
- add description text as tool tip to name label in drop-down dialog (discoverability might be low; difficult to design for touch interface (long-press is even more difficult to discover + item already has a touch area))
- add (context) help (e.g. via icon) to the title bar of the drop-down dialog (opening a pop-up on top of another (drop-down) dialog is considered bad style)
- add (context) help (e.g. via icon) to the create component screen for the meter type parameter (description and input possibility is separated --> contradicts the 'recognition rather than recall' paradigm)

If UI does not need to be touch controlled, tool tip might be preferred solution. *Hint:* context help concept could be a good idea not only for this case.

I also found another file containing briefer information regarding the meter type: https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.io.shelly/src/io/openems/edge/io/shelly/shellypro3em/Config.java I'm not sure where this file is used, but maybe it's worth to expand the description there with the javadoc from the first file mentioned above as well.
 
Screenshot, showing meter type drop-down dialog:
[image]

sfeilmeier, 2025-08-03:
We are aware, that the `MeterType` is not very expressive the way it is now. It is also not well defined what exactly happens in which configuration. 

Apart from that the "Add Component" dialog is very technical by definition. Here it's not possible to add more context to the configuration items. This is only possible via App Center - see SOCOMEC-Meter as an example: https://github.com/OpenEMS/openems/blob/develop/io.openems.edge.core/src/io/openems/edge/app/meter/SocomecMeter.java

I'll close here, because there are multiple ways to improve this, e.g.
- properly define/describe MeterTypes somewhere in the OpenEMS docs
- add Shelly Apps for App Center and add descriptive text there

closed 2025-08-03 by sfeilmeier

sjjh, 2025-08-03:
> I'll close here, because there are multiple ways to improve this, e.g.
I'm probably neither experienced enough to evaluate the best way, nor to implement it. Do you open another issue to track it?