OpenEMS/openems #3214: influxdb is not reachable from within edge or ui docker container

sjjh, 2025-06-25:
### Description

I've followed the getting started guide to setup OpenEMS Edge, UI, and InfluxDB via docker. I'm using following `docker-compose.yml`:
```services:
  openems-edge:
    #image: openems/edge:latest
    image: openems/edge:develop
    container_name: openems_edge
    hostname: openems_edge
    restart: unless-stopped
    volumes:
      - openems-edge-conf:/var/opt/openems/config:rw
      - openems-edge-data:/var/opt/openems/data:rw
    ports:
      - 8080:8080 # Apache-Felix
      - 8085:8085 # UI-Websocket

  openems-ui:
    image: openems/ui-edge:latest
    container_name: openems_ui
    hostname: openems_ui
    restart: unless-stopped
    volumes:
      - openems-ui-conf:/etc/nginx:rw
      - openems-ui-log:/var/log/nginx:rw
    environment:
      - UI_WEBSOCKET=ws://test-openems.schloss-tempelhof.de:8085
    ports:
      - 80:80
      - 443:443

  openems_influxdb:
    image: influxdb:alpine
    container_name: openems_influxdb
    hostname: openems_influxdb
    restart: unless-stopped
    volumes:
      - openems-influxdb:/var/lib/influxdb2:rw
    ports:
      - 8086:8086

volumes:
  openems-edge-conf:
  openems-edge-data:
  openems-ui-conf:
  openems-ui-log:
  openems-influxdb:
```
After having configured InfuxDB in the UI, the component is displayed working/green, but it actually fails to connect to InfluxDB. The debug log within the UI shows: 
```
InfluxDB read is temporarily blocked [0.950].
25.6.2025, 23:47:05
WARN
io.openems.common.websocket.OnMessageHandler
[ctrlApiWebsocket0] JSON-RPC Error "Failed to connect to localhost/[0:0:0:0:0:0:0:1]:8086" for Request {"method":"edgeRpc","params":{"edgeId":"0","payload":{"method":"queryHistoricTimeseriesEnergy","params":{"timezone":"Europe/Berlin","fromDate":"2025-06-25","toDate":"2025-06-25","channels":["_sum/C...
25.6.2025, 23:47:05
ERROR
io.openems.edge.common.jsonapi.JsonApiBuilder
Failed to connect to localhost/[0:0:0:0:0:0:0:1]:8086
25.6.2025, 23:47:05
ERROR
io.openems.shared.influxdb.proxy.InfluxQlProxy
InfluxDB query runtime error. Query: SELECT LAST("_sum/ConsumptionActiveEnergy") - FIRST("_sum/ConsumptionActiveEnergy") AS "_sum/ConsumptionActiveEnergy", LAST("_sum/GridBuyActiveEnergy") - FIRST("_sum/GridBuyActiveEnergy") AS "_sum/GridBuyActiveEnergy", LAST("_sum/GridSellActiveEnergy") - FIRST("_sum/GridSellActiveEnergy") AS "_sum/GridSellActiveEnergy", LAST("_sum/ProductionActiveEnergy") - FIRST("_sum/ProductionActiveEnergy") AS "_sum/ProductionActiveEnergy" FROM data WHERE time > 1750802400s AND time < 1750888800s, Error: Failed to connect to localhost/[0:0:0:0:0:0:0:1]:8086
```
I can `ping` the InfuxDB from the host, as well from within the InfluxDB docker container, but not from within the Edge or UI docker container:
```
root@Test-OpenEMS:~# curl -sL -l -v localhost:8086/ping
*   Trying 127.0.0.1:8086...
* Connected to localhost (127.0.0.1) port 8086 (#0)
> GET /ping HTTP/1.1
> Host: localhost:8086
> User-Agent: curl/7.88.1
> Accept: */*
> 
< HTTP/1.1 204 No Content
< Vary: Accept-Encoding
< X-Influxdb-Build: OSS
< X-Influxdb-Version: v2.7.12
< Date: Wed, 25 Jun 2025 21:36:22 GMT
< 
* Connection #0 to host localhost left intact
root@Test-OpenEMS:~# docker exec openems_influxdb curl -sL -l -v localhost:8086/ping
* Host localhost:8086 was resolved.
* IPv6: ::1
* IPv4: 127.0.0.1
*   Trying [::1]:8086...
* Connected to localhost (::1) port 8086
* using HTTP/1.x
> GET /ping HTTP/1.1
> Host: localhost:8086
> User-Agent: curl/8.12.1
> Accept: */*
> 
< HTTP/1.1 204 No Content
< Vary: Accept-Encoding
< X-Influxdb-Build: OSS
< X-Influxdb-Version: v2.7.12
< Date: Wed, 25 Jun 2025 21:36:51 GMT
< 
* Connection #0 to host localhost left intact
root@Test-OpenEMS:~# docker exec openems_ui curl -sL -l -v localhost:8086/ping
* Host localhost:8086 was resolved.
* IPv6: ::1
* IPv4: 127.0.0.1
*   Trying [::1]:8086...
* connect to ::1 port 8086 from ::1 port 59856 failed: Connection refused
*   Trying 127.0.0.1:8086...
* connect to 127.0.0.1 port 8086 from 127.0.0.1 port 41444 failed: Connection refused
* Failed to connect to localhost port 8086 after 0 ms: Could not connect to server
* closing connection #0
root@Test-OpenEMS:~# docker exec openems_edge curl -sL -l -v localhost:8086/ping
* Host localhost:8086 was resolved.
* IPv6: ::1
* IPv4: 127.0.0.1
*   Trying [::1]:8086...
* connect to ::1 port 8086 from ::1 port 41746 failed: Connection refused
*   Trying 127.0.0.1:8086...
* connect to 127.0.0.1 port 8086 from 127.0.0.1 port 33902 failed: Connection refused
* Failed to connect to localhost port 8086 after 0 ms: Could not connect to server
* closing connection #0
```

### Screenshots

[image]

[image]

[image]

### Operating System

Debian 12 vm

### How to reproduce the Error?

1. Use the getting started docker guide to install Edge & UI
2. Add InfluxDB as described in getting started docker guide
3. Check system to encounter error

sjjh, 2025-06-25:
@da-Kai probably that's something for you to look into?

da-Kai, 2025-06-27:
@sjjh
This is not an OpenEMS issue but rather a general Docker networking question. Please post these kinds of questions in the [Openems Community](https://community.openems.io/) instead of creating a GitHub issue here.

Note that you cannot use localhost inside a Docker container to reach other containers. Instead, you need to use the container hostnames you defined in your docker-compose.yml, such as openems_influxdb.

If this simple change doesn't work you, also specify a shared docker network.
see https://docs.docker.com/compose/how-tos/networking/#specify-custom-networks

sjjh, 2025-06-27:
My `docker-compose.yml` does not contain any reference to `localhost` (this was just a naive try to debug the issue). As I basically was just following the *getting started guide*, I thought it might be a bug within the guide.

sjjh, 2025-06-27:
I do see, that the [getting started guide](https://openems.github.io/openems.io/openems/latest/backend/deploy/docker.html) correctly specifies the container name instead of `localhost`. After correcting my setup, it now works. Thanks for your support!